openAVproductions/openAV-Ctlra #1: Support for non-USB backends

harryhaaren, 2017-05-17:
Currently the Ctlra API is pretty USB centric, particularly in the `accept` function, where only the `vendor` and `device` integers representing USB vendor and device are available. This should probably have an enumerator that state the backend-type of the device, and allow the application to check what type of device, and then the actual device metadata.

Be-ing, 2017-06-24:
Mixxx uses a [udev rule](https://github.com/mixxxdj/mixxx/blob/master/res/linux/mixxx.usb.rules) that whitelists known USB vendor IDs.

Be-ing, 2017-06-24:
[hidapi](https://github.com/signal11/hidapi) which Mixxx uses supports HID over Bluetooth as well as USB and works on GNU/Linux, Mac, and Windows.

harryhaaren, 2017-06-24:
HIDAPI uses a pretty strange thread-model, where each device has its own thread. Currently Ctlra doesn't spawn any threads - keeps things simpler. I'm not 100% against using threads if necessary... but libUSB's async mode works fine so far. If there is a PR for a device that requires HIDAPI, I'll re-think :)

Be-ing, 2017-06-24:
There won't be a PR for a Bluetooth HID device until the library supports Bluetooth HID ;)

harryhaaren, 2017-06-24:
Are there even any common BlueTooth HID devices? WiiMote uses its own protocol IIRC.. and a very quick google doesn't show BlueTooth HID there. WiiMote is about the most common bluetooth device I know of... Anyway - there's lots of work to be done, starting with supporting more "real" devices that actual DJs own! :)

Be-ing, 2017-06-24:
I agree there are more important tasks to focus on now, but wireless controllers are becoming more common. The [Mixfader](http://djtechtools.com/2015/03/17/the-mixfader-a-bluetooth-enabled-crossfader/) uses Bluetooth but I'm not sure if it uses HID or MIDI over Bluetooth. The [Akai LPD8 Wireless](http://www.akaipro.com/product/lpd8-wireless), [Livid Minim](http://lividinstruments.com/products/minim/), and [Roli BLOCKS](https://roli.com/products/blocks/) use MIDI over Bluetooth.

Be-ing, 2017-06-24:
Oh, good news! ALSA now [supports MIDI over Bluetooth LE](https://blog.felipetonello.com/2017/01/13/midi-over-bluetooth-low-energy-on-linux-finally-accepted/)! So any backend that interacts with ALSA's MIDI support will work for Bluetooth MIDI controllers. For cross platform MIDI support, [PortMIDI](http://portmedia.sourceforge.net/portmidi/) could be used, but AFAIK it does not support hotplug... yet...

Be-ing, 2017-06-24:
I think [RtMidi](https://github.com/thestk/rtmidi), another cross platform MIDI library, supports hotplug. It was [considered for use in Mixxx in 2013](https://blueprints.launchpad.net/mixxx/+spec/rtmidi-backend) and decided against at the time. I don't know if the library has changed since then.

Be-ing, 2017-06-24:
I think using cross platform libraries that abstract the communication protocol (HID, MIDI, OSC, or whatever) away from the wire(less) protocol would be a wiser idea than using libusb directly.

harryhaaren, 2017-06-24:
Ultimately these things will be decided by PRs not discussion. I don't like the idea of bringing in C++ code, because right now Ctlra is just C. This simplifies creating bindings to other langauges, and enables easy compilation on embedded platforms. The ALSA MIDI backend serves its purpose well - I like choices that are made by showcasing code that requires things to be a specific way.

If a PR comes in requesting library or dependency X, and it upholds some technical scrutiny, then absolutely yes lets merge it - but lets wait until these issues arise in real life. Attempting to preempt things without technical insight is not a good solution.

Thanks for the links to the controllers - I wasn't familiar with some of them, and they do seem interesting!

Be-ing, 2017-06-24:
> Ultimately these things will be decided by PRs not discussion.|

Sure, but it's good to have an idea of what direction to go with development before diving into code.

> I don't like the idea of bringing in C++ code, because right now Ctlra is just C.

Understandable. RtMidi is C++, but PortMIDI is C.

> Thanks for the links to the controllers - I wasn't familiar with some of them, and they do seem interesting!

I don't know how much relevance this has to Ctlra, but you may want to take a look at the [Novation Launchpad Pro firmware](https://github.com/dvhdr/launchpad-pro).

simonvanderveldt, 2017-07-03:
@harryhaaren would this be the issue to determine if and if so how to add MIDI support to ctlra or should I create a new issue similar to the OSC one?

harryhaaren, 2017-07-03:
Hey Simon, Lets create a new issue - better to keep things split. Feel free to open a ticket, we can link back here as the root of the conversation :)

harryhaaren, 2018-01-06:
ALSA MIDI backend is now supported via `midi_generic.c` device file, closing.

closed 2018-01-06 by harryhaaren