openAVproductions/openAV-Ctlra #2: Collaboration?

jcelerier, 2017-05-18:
Hi!

Just watching your conference @LAC; I think that it could be interesting to work together. I'm working on the [libossia](https://github.com/OSSIA/libossia) library which aims to bring a similar kind of abstraction of hardware / software peripherals to end-user software (but started from OSC, Midi, and other high-level description protocols instead of USB HID & friends).

It's written in C++ but has a (small, unfinished) C wrapper, and has ports to multiple environments. 

Best,
Jean-Michaël

harryhaaren, 2017-05-21:
Hi,

I've had a look at libossia, and it looks like a good candidate to consume the Ctlra library and utilize it in order to enable hardware interfaces integrate with the existing libossia software. If you have any questions on the current Ctlra API, do ask :)

Cheers, -Harry

jcelerier, 2017-05-21:
cool, thanks :)

closed 2017-05-21 by jcelerier