openAVproductions/openAV-Ctlra #4: Enable non-root user access

harryhaaren, 2017-05-18:
Ensure that a good UX is provided for users who are not running as root. This probably involves writing udev rules to enable non-root user access to the USB devices that are supported by ctlra.

harryhaaren, 2017-05-18:
Thanks Robin for mentioning Ardour's udev rule for the machine2:
https://github.com/Ardour/ardour/blob/850b0d3e0940df07edaafed30e6517ec3f6f309c/tools/udev/99-maschine2.rules

Be-ing, 2017-06-24:
Mixxx uses a [udev rule](https://github.com/mixxxdj/mixxx/blob/master/res/linux/mixxx.usb.rules) that whitelists known USB vendor IDs.

harryhaaren, 2017-06-24:
Ah thanks for the input here - Udev rules it is, Thanks for the note to investigate the existing rules that Mixxx puts in place for USBHID controller devices .

Be-ing, 2017-06-24:
Ableton's USB vendor ID is [apparently 2982](https://github.com/Ableton/push2-display-with-juce/issues/1).