openAVproductions/openAV-Ctlra #6: Dummy device

foolswood, 2017-05-20:
For testing integrations without hardware :)

harryhaaren, 2017-05-20:
Thanks for noting this here. Yes absolutely agree this would be helpful - and essential to enable developers easily integrate Ctlra. Designing some UI that allows flexibility and clicking items, showing LED feedback would be a longer term goal.

harryhaaren, 2017-05-22:
Update, prototype UI that sends Ctlra events on button clicks is implemented. The UI uses the AVTK library, and a button click event results in the application recieving a Ctlra event. The code is currently very static (hardcoded linker paths etc) so I can't push it yet, but its on the way :)

harryhaaren, 2017-05-22:
Note also that in order to properly support "virtual" devices (aka, not real hardware) we must also fix #1, to allow the application to identify what is a virtual device, and what is "real". Use-case wise, a virtual device can always exist - hence the application has to be able to decide to accept/reject based on some conditions.

foolswood, 2017-05-23:
Shiny!

harryhaaren, 2017-07-09:
@foolswood -  how would you (as an application) like to specify that Ctlra should provide a dummy device? Keep in mind there may be more than one type of dummy device, so some mechanism to request a specific one would be required...

harryhaaren, 2017-08-22:
Update; i'm working on an API that allows the application to call `ctlra_dev_virtualize`, which will take a `ctlra_dev_info_t` and use that. In short, it can pass in the device it wishes to have a virtual copy of. I'll post commits soon when its in a reasonable state to review.

foolswood, 2017-08-23:
Sorry I missed the previous comment, your solution seems sensible.
What I'm looking for at the moment is a way to get ctlra to push some event so I can check my handling code in a unit testable way.

harryhaaren, 2017-09-17:
Please see pull request https://github.com/openAVproductions/openAV-Ctlra/pull/41 for details on virtualizing hardware devices. It is pretty much all available right now - there's a few rough edges, and device discovery needs some updates / maintenance to become properly useful. The new sample application "device_test" showcases the integration of HW and SW if you have a Ctlra supported device now.

harryhaaren, 2017-10-02:
PR #41 has been merged - so there are now "virtual devices" available in the Ctlra master branch. The following command emulates the S2 device, each device is identified by its "Vendor" "Device" string pairs (Quoted here, as they contain spaces .. ):
```./virt_dev "Native Instruments" "Kontrol S2 Mk2"```
```./virt_dev``` on its own defaults to the Z1 (smaller easier-to-fit-on-a-screen little device)

Gonna close this issue, as you now have multiple "dummy" (or "real") devices :)

closed 2017-10-02 by harryhaaren