openAVproductions/openAV-Ctlra #7: [Feature] Optionally provide device layout information

fundamental, 2017-05-27:
This was mentioned at LAC, but I figured it was a good idea to create a feature request out of it.

I would like to request that if this library supports a number of more complex complex controllers, then it may also supply layout information about their front panel. At the moment it seems simple to provide a mapping of "controller X is at 'a' mm from the top, 'b' mm from the left side and is 'c' x 'd'" with the assumption that all devices can be roughly discussed in terms of one 2D front panel with a known metric size. 

_Why would this be useful?_ Well, for mapping physical controls in software. IMO users will want to be able to see what the device's controls are hooked up to in a geometric sense (i.e. a picture showing the top left knob goes to volume, the one to the right of it goes to bpm, etc) rather than in a list of "controller 1 - volume, controller 23 - bpm". Providing information about where controls are physically located also makes it simple to create mirrored/software versions of the device for virtual interaction (testing/applications showing users information about the device/etc).

_How would this information be presented?_ I'm guessing that a simple representation would be best including:
- front panel size (in mm)
- a list of physical controllers
- an xywh position for each (in mm)
- a type of controller for each (may be redundant) (knob/slider/led/display/etc)

harryhaaren, 2017-05-31:
Hey, Thanks for filing the issue here - good to keep track of everything centrally. I do see some value in exposing the info, however there is quite some workload to actually expose the info per device. Note that I'm not opposed to it, just noting that its work - and should be prioritized..!

The list of input physical controls, and the type of each input control is already exposed in the ctlra API, so that info is available. Output LEDs are currently enumerated in the device specific header files, however it would be good to provide some more metadata / higher-level API for the output side... *thinking in progress*

The work here would be to expose the "whole" device size, and measure XYWH co-ordinates in mm and expose that per item, and ensure that the output API is sane from the application POV.

Thanks again for noting it here! -Harry

Be-ing, 2017-06-24:
Another way this could be accomplished could be mapping Ctrla events to objects in an SVG.

fundamental, 2017-06-24:
I would strongly recommend against SVG in the ctlra interface as it would introduce a large amount of complexity making this functionality much less likely to be both implemented and used IMO.

harryhaaren, 2017-06-24:
Given its a C API, I'd like to provide native C datatype info. I've mocked up an API what it would look like to expose the control location data, I'll clean up some details and post it as a branch for opinions.

Having the control names etc expose also relates to https://github.com/openAVproductions/openAV-Ctlra/issues/6, and providing a "virtual" device with co-ordinates leads nicely to virutalizing *actual* devices. More details to follow with the code later! :)

harryhaaren, 2017-09-15:
Virtual Device branch contains layout info on a few of the devices. The infrastructure and API have been updated, statically exposing a variety of Ctlra Item objects, which can be read to build up a model of any controller - even if it is not physically present. For details, see the following branch, which will be merged into master soon-ish. https://github.com/openAVproductions/openAV-Ctlra/tree/virtual_device

closed 2017-09-15 by harryhaaren