openAVproductions/openAV-Ctlra #11: clarify the goals of Ctrla

Be-ing, 2017-06-24:
After watching the [recording of the presentation at LAC](https://www.youtube.com/watch?v=-bwynA7HRow) I think I have a better understanding of the goals of Ctrla. Please correct me if I am misunderstanding.

1. Parse HID packets to determine which parts have changed state from the last packet. This turns working with HID controllers into more like working with MIDI controllers.
2. Provide a programming environment to map the events from layer 1 to manipulate the state of the host application. The mapping would also react to state change events from the application to send output to the controller.

Does one library need to do both? Would it make more sense to leave it at step 1 and provide an API to pass the events in and out of host application's scripting environment? Or maybe split Ctrla into two libraries that are intended to be used together but don't necessarily have to be used together?

If Ctrla aims to do step 2, as discussed previously on IRC, I think it should provide one more layer of abstraction analogous to the [Components JS](https://mixxx.org/wiki/doku.php/components_js) library for Mixxx. This groups handling of input and output events for a physical component of the controller (knob, fader, button, or encoder) into an object to make it easy to share state between the input and output handling. Instead of hardcoding the mapping between input/output events to the state of something in the host application like Mixxx's XML MIDI mapping system, the part of the application that the Component JavaScript object interacts with is treated as mutable state (specified by the "inKey", "outKey", and "group" properties). This makes it easy group Components into a container object to iterate over and change the state of a group of Components to switch between multiple layers of functionality. For more complex input/output handling, the inKey/outKey state can be disregarded and a custom function can be implemented. Working at this level of abstraction has made it relatively easy to support really complex mapping designs, particularly with the [EffectUnit object](https://mixxx.org/wiki/doku.php/components_js#effectunit) provided by the library. I think a good test for the usefulness of Ctrla would be how easy it is to reimplement that same design for the Kontrol X1 Mk2, Kontrol S4 Mk2, and Kontrol S2 with code that can be easily shared between the mappings for each device.

As far as I can tell the approach taken by the Components library is novel. I think Ableton's Python API used for the Push might be somewhat similar (from @arximboldi who worked on the team that designed the Push) but I have not found any public documentation for it. Providing this approach to controller mapping to applications would make Ctrla very useful. I think it would make it worthwhile to implement MIDI support into Ctrla which could potentially replace the ad-hoc MIDI controller support various applications have implemented.

Be-ing, 2017-06-24:
If Ctrla aims to provide the mapping programming environment, I think user preferences for mappings (issue #10) should be in the scope of Ctrla.

Be-ing, 2017-06-24:
It may be worthwhile to consider designing Ctrla so that mapping authors have a choice of programming language. For example, TCC, Python, Lua, or Node could all be used.

harryhaaren, 2017-06-24:
Ctlra is the "lowest part", where hardware device specifics are abstracted into generic events for consumption by the application. I do not see the mapping as a "core" part of Ctlra, the TCC example merely shows-off this opportunity.

I do see a gap between the generic events (the "top" of the Ctlra library) and the application functionality APIs. In short, we need a "event to application mapping" library. This point was raised by a few developers right after the Ctlra talk: accessing the device is one thing, but providing a mechanism to map that to the application in question is the harder part.

Ctlra solves HW device abstraction, and seems to do so well. I don't feel Ctlra should enforce any mapping strategy on the Application. 

I'm wary of "scope-creep" where the library becomes too wide-spread, and attempts to achieve too much. MIDI backwards compatibility is on the fence here - i see value, but it is not really the core of the Ctlra concept.

I'm open to language bindings, particularly I'd like to see a Python binding due to Python's awesomeness for fast prototyping and generally has an easy-to-use API for everything.

</rant> there's a lot of things to discuss / improve in this domain, but ultimatlely achieving one thing with the Ctlra library would provide value - mushing more features in and failing would not, hence my careful concideration of "feature requests" as such.

Be-ing, 2017-06-24:
Okay, thanks for clarifying. I think a library just for abstracting device interaction into events is useful and a good step forward. A separate Ctrla-event-to-application mapping library on top of that would be great too, but one step at a time. :)

> MIDI backwards compatibility is on the fence here - i see value, but it is not really the core of the Ctlra concept.

I think MIDI compatibility is important for this library to be really useful. If you were starting a new application, would you want to support both Ctlra and ALSA sequencer/PortMIDI/RtMidi, or just implement Ctlra support and be done with the application layer to communicate with devices?

Btw, I find it difficult to not type "Ctrla" instead of "Ctlra". The former is more intuitive because it looks and sounds like "controller".

harryhaaren, 2017-06-24:
RE Ctlra.. you'll get used to it.. also in that case, I'll fix the title ;)

Yes that is the value I see too, but is that worth "draggin in" the whole MIDI spec for? Or attempting to build "MIDI v2" or something? There have been attemps (OSC, LV2 Atoms, ... ) to "replace" MIDI with something better, and ultimately most DAWs just use plain-old MIDI. I don't see Ctlra changing that in the short or medium term.

Ctlra won't answer all of the issues of supporting devices, just the lowest part - abstracting away hardware. The rest can build on top of that - exactly as you say "one step at a time"!

fundamental, 2017-06-24:
> would you want to support both Ctlra and ALSA sequencer/PortMIDI/RtMidi

This gets into slippery territory. What about having Ctlra supporting virtual midi through JACK? what about OSS? what about windows MIDI options? and coreaudio specifics?

Along these lines however one question I have about scope is: should Ctlra support simpler devices which have an API based around MIDI CCs + a few under-documented sysex control messages? (e.g. the nanokontrol series of controllers). It's reasonable from my view for Ctlra to be a new abstraction over any physical DAW/MIDI controller or for it to only target complex controllers which aren't fully usable without custom drivers, though I think the goal should be clarified at some stage.

harryhaaren, 2017-06-24:
@fundamental ; fair point on abstracting MIDI controllers, and parsing them into Ctlra events. And this is something that I do see in Ctlra. Abstracting "MIDI" out of a DAW is a step further - and that gets a bit harder.

I like ALSA MIDI because it is supported by the kernel, and that (usually) means that the device is a physical device attached to your computer. With JACK MIDI, that is usually not the case - because SW applications use JACK MIDI (and a2jmidid and jack1 can transport the ALSA ports to JACK MIDI ports by shipping events between the two APIs).

Note that no USB device is going to ever have a JACK MIDI *native* backend - because JACK MIDI is not in the kernel, and hardware tends to be plugged into the kernel. AKA: ALSA MIDI is all we should ever need. Adding JACK MIDI would add complexity (particuarly in threading) and a dependency on JACK. I like JACK - but Ctlra is not just for JACK applications.

TL;DR: Enabling MIDI controller hardware under Ctlra - yes absolutely. Re-inventing MIDI inside the Ctlra Events structure - I currently think this is not a good solution.

Be-ing, 2017-06-24:
> Yes that is the value I see too, but is that worth "draggin in" the whole MIDI spec for? Or attempting to build "MIDI v2" or something? There have been attemps (OSC, LV2 Atoms, ... ) to "replace" [...]

The difficulty with trying to replace MIDI with a more appropriate protocol for today's fancier devices is that MIDI has a ton of inertia behind it. Even though Native Instruments is using HID for their own controllers, Traktor still has to support MIDI and NI built a MIDI translation layer into their Mac and Windows drivers so their hardware could work with other software. Perhaps in the long term Ctlra could allow device manufacturers to use whatever device communication protocol they think is appropriate without application developers -- both the developers of application the device manufacturer is designing for and third party developers -- having to implement and maintain separate implementations for MIDI and HID or whatever other protocol a device manufacturer decides to use.

> Note that no USB device is going to ever have a JACK MIDI native backend - because JACK MIDI is not in the kernel, and hardware tends to be plugged into the kernel.

Good point. I think supporting JACK MIDI should therefore stay out of the scope of Ctlra.

Be-ing, 2017-06-25:
Regarding using ALSA MIDI directly, realistically, I don't foresee any device manufacturer thinking it would be a good use of their employees' time to contribute to Ctlra if it does not work on Mac and Windows with no extra effort for the device manufacturer or application developer. If it stays Linux only, it's just another obscure tool Linux audio nerds like us, and the burden will remain on us to add Ctlra support for every new device. On the other hand, if it is made useful to device manufacturers, and they contribute Ctlra support for their devices, then we benefit from easier support of their devices.

Be-ing, 2017-06-25:
The more I think about, the less sure I am that Ctlra's scope should include MIDI controllers. It should probably focus on support HID well first, then the question could be revisited. Regardless, whatever communication protocols are supported, they should be supported by using cross platform libraries.

harryhaaren, 2017-06-25:
Keep in mind that there are multiple ways to achieve "cross-platform-ness". One (perhaps the easy choice) is to use a pre-canned x-platform library, and depend on that. Another solution is to provide an API for a feature eg; get_midi_message(), and implement that natively per OS. This approach reduces dependencies, and allows the workings of the x-platform abstraction to be modified if required.

Small libraries from the LV2 world take approach 2 providing the low-level x-platform implementations themselves, a good example is [PUGL](https://drobilla.net/software/pugl).

LibUSB is cross-platform in the sense of option 1), so hopefully this "just works". MIDI on the other hand, I think might be better to take option 2, in order to remove dependencies and attempt "direct connecting" to the device without a patchbay inbetween.

Be-ing, 2017-06-25:
Sure, you do not have to use an external library to implement cross platform support. But many other developers have already faced the same task and so we have libraries that take care of the hard parts. I understand the desire to minimize external dependencies, but I think depending on small plain C libraries like libUSB and PortMIDI is not a big deal. It's not like depending on a large framework like GTK or Qt (although Qt is more modular now and can be used only for specific parts).

harryhaaren, 2017-07-09:
Be-ing said:
> The more I think about, the less sure I am that Ctlra's scope should include MIDI controllers. It should probably focus on supporting HID well first, then the question could be revisited

Good idea. I'll close this - I think the recent discussions on the tracker here, as well as on the Mixxx-devel lists have solidified what the scope of Ctlra is. If there are any follow up questions, we can always open new issues :)

closed 2017-07-09 by harryhaaren