openAVproductions/openAV-Ctlra #9: license?

Be-ing, 2017-06-24:
There is no license in the code repository. Have you thought about what license to use for this? I think it's worth considering the LGPL. It's also worth [considering the GPL](https://www.gnu.org/licenses/why-not-lgpl.html).

fundamental, 2017-06-24:
Please take a look at the code files themselves as they appear to be properly licensed with a 3-clause BSD license (e.g. https://github.com/openAVproductions/openAV-Ctlra/blob/master/ctlra/event.h )

harryhaaren, 2017-06-24:
Yep BSD is the plan (as per code files) because it would allow prop software to integrate Ctlra. From a HW manufacturer's POV I hope this encourages using Ctlra in applications :)

I'll leave this item only to add to the README what the license is, and have it called out more clearly. Thanks for noting that.

Be-ing, 2017-06-24:
> From a HW manufacturer's POV I hope this encourages using Ctlra in applications :)

Thinking about this more, that would provide a benefit for free software. If hardware manufacturers implement support for their device in Ctrla it will be easier for free software to interact with that device.

> I'll leave this item only to add to the README what the license is, and have it called out more clearly. Thanks for noting that.

It would help to have a LICENSE file too.

harryhaaren, 2017-06-24:
Yep exactly. Agreed, I actually meant LICENSE file instead of README xD thanks!

Be-ing, 2017-06-24:
The source code files contain "All rights reserved" above the license permissions. Those are contradictory. If you want this library to be used by device manufacturers I would not be surprised if their lawyers nitpick about that.

Be-ing, 2017-06-24:
I think the LGPL would be a better choice of license. With a lax permissive license, device manufactures could make proprietary modifications to Ctlra for their devices and bundle it with proprietary applications without others being able to benefit from their use of Ctlra.

foolswood, 2017-06-25:
As I understand it the BSD license is less restrictive than the LGPL. The LGPL isn't "infectious" (i.e. requiring that code linking to the library be under a similar copyleft license) like the GPL but IIRC it does require changes to the project itself to be shared. However BSD licenses doesn't usually even stipulate that.

harryhaaren, 2017-06-25:
The industry is weary of anything GPL, due to its "viral" component, and that if they accidentally ship a binary with LGPL statically linked, they by right have to share all the code compiled in.

See LGPL has to be linked as a shared object in order to not "infect" the application. Also correct on changes to the LGPL code have to be shared.

BSD is very permissive - and hence should allow for adoption more. Agree that there is the possibility of applications using Ctlra and not contributing back, but the value of a 3rd party app adopting Ctlra is IMO still there, as it means Ctlra becomes more used.

Be-ing, 2017-06-25:
> if they accidentally ship a binary with LGPL statically linked, they by right have to share all the code compiled in. See LGPL has to be linked as a shared object in order to not "infect" the application.

The purpose of this restriction of the LGPL is to allow users to use modified versions of the library with proprietary applications. If the scope of Ctlra is confined to abstracting the details of how the device uses HID or another communication protocol as discussed in #11, IMO there is little value in allowing users of proprietary programs to modify the version of Ctlra used by the proprietary program. In this case, the [Mozilla Public License](https://www.mozilla.org/en-US/MPL/2.0/) would be a better choice than the LGPL.

> the value of a 3rd party app adopting Ctlra is IMO still there, as it means Ctlra becomes more used.

I understand how that could be rewarding for you as the developer of Ctlra, but for the wider community I do not think there is much benefit to a proprietary application using Ctlra if others cannot reuse the modifications the proprietary application developer made to Ctlra to support specific devices.

harryhaaren, 2017-07-09:
I see good reason to use BSD:
1) Imagine a company wants to beta-test an unannounced controller, using various "trusted-members" of the community to get feedback. The LGPL/MPL would demand that their Ctlra driver code would also be posted when they "distribute" the software to members of the community: that is a barrier to usage for any company. (The Ctlra driver would showcase the capabilities of the device before it is announced - that's not gonna fly).
2) Currently all the code is BSD licensed. Changing the license would require a written statement from all contributors that they are OK with changing the license to a more-restrictive one. Of course this "offical" letter should be reviewed by legal etc etc - just not gonna happen.

So all in all, I don't see Ctlra changing from BSD. I do understand your concern that it could be used and nothing returned - and your concerns are valid. Ultimately I believe that companies would like to contribute to communities - after all it is the community of people that buy the gear that keep the company in business. I hope any company using Ctlra does contribute their improvements back, I think it will enable creatives to utilize their hardware for fantastic things.

I'll mark this issue as closed in a few days if things are quiet. -Harry

Be-ing, 2017-07-10:
> The LGPL/MPL would demand that their Ctlra driver code would also be posted when they "distribute" the software to members of the community

It would not require that. It would require that the source code is made available to those developers, who could legally post it online. If a company wanted to do something like this, IMO they should hire the people they want feedback from. Asking experts for their opinion for free for the company's commercial benefit would not be fair. If they were hired as contractors, maybe they would have the power to keep it unpublished, but IANAL, and that might depend on jurisdiction. As far as I understand, they could legally keep the code unpublished as long as their own employees were the only ones using it.

> Changing the license would require a written statement from all contributors that they are OK with changing the license to a more-restrictive one.

There are only 3 contributors so far, so that wouldn't be difficult if they all agree.

> Ultimately I believe that companies would like to contribute to communities - after all it is the community of people that buy the gear that keep the company in business.

I wish that we lived in a world where we could rely on that, but we don't, which is why copyleft exists.

PR #31 (2017-08-07): Create LICENSE.txt file
Add LICENSE file, close #9.
commit ce83c016 (2017-08-07): Create LICENSE.txt file

Add LICENSE file, close #9.
closed 2017-08-07 by harryhaaren