openAVproductions/openAV-Ctlra #13: Mixxx integration

Be-ing, 2017-06-25:
How will Ctlra integrate with Mixxx? @harryhaaren I know you've been working with Mixxx to develop Ctlra but I am curious to hear more details. I could imagine a new Controller subclass for Ctlra that integrates with Mixxx's JavaScript environment. I think the Ctlra Controller subclass would only need to expose a few methods to the JS interpreter with the Q_INVOKABLE macro:
  * makeInputHandler: this would work like my [proposal for registering MIDI input handlers from JS](https://mixxx.org/wiki/doku.php/registering_midi_input_handlers_from_javascript). Instead of an array of MIDI bytes as the first argument to makeInputHandler, it would take some identifier that links the handler to a Ctlra event. This would return an object to the script with a `disconnect` method and a UUID.
  * send: send a Ctlra output event to the controller
  * freeze/unfreeze: When the freeze method is called, suspend sending output packets to the HID device until unfreeze is called. This would allow the script to "send" multiple output events to the controller simultaneously with one HID packet, which would be useful for switching between layers of functionality and seeing the state change on the device all in one instant. The common-hid-packet-parser.js script supports this way of working.

The Components library could be adapted to wrap the input handlers from makeInputHandler and engine.makeConnection into JS objects for each component of the device.

Be-ing, 2017-06-25:
Essentially, this approach is what I have been thinking about scaling down the common-hid-packet-parser.js library to do, except that other applications will be able to benefit from abstracting the details of how the hardware uses HID.

harryhaaren, 2017-06-25:
Yes correct I have been working on Ctlra and Mixxx integration - particularly to test the Ctlra API for usability, and to ensure that the correct APIs are available at the correct times.

Regarding Ctlra implementation in Mixxx, I'm working on refactoring it to be maintainable and not a proof-of-concept hacky implementation. Ctlra will reside inside the ControllerManager, and *not* as a subclass of a Controller. The reasoning is that one Ctlra instance handles multiple Controllers - so its not the right abstraction to derive from.

Regarding mapping of Events to Mixxx, keep in mind that the Ctlra API is plain C, and Mixxx provides the ControlObject to interact with the Mixxx core. Using TCC we can script the event bindings in C, which provides all the functionality required. That is currently the only work planned.

Be-ing, 2017-06-25:
> one Ctlra instance handles multiple Controllers

While it makes sense for one Ctlra instance to handle multiple controllers, I think Ctlra should expose an API to enumerate attached Ctlra compatible devices and enable/disable them. Ctlra should handle hotplug in the sense of detecting when devices are plugged in and unplugged, but I think it should be up to the application to decide whether to use the device so the application can provide a user interface for this. A user may not want to use a particular device with a particular application even if there is a Ctlra compatible mapping for it. Requiring the user to unplug a device in such a situation would be annoying.

> Regarding mapping of Events to Mixxx, keep in mind that the Ctlra API is plain C, and Mixxx provides the ControlObject to interact with the Mixxx core. Using TCC we can script the event bindings in C, which provides all the functionality required. That is currently the only work planned.

If the scope of Ctlra is restricted to abstracting the device details into events as discussed in #11, I think it would make more sense to integrate Ctlra into applications' existing scripting environments. With the TCC approach, a Mixxx controller mapping developer has to learn a different programming language and API for interacting with Mixxx depending on whether they are mapping a MIDI or Ctlra controller. Also, the application developer has to do the work of exposing a way to interact with the application to both Ctlra and their existing MIDI mapping system. in the case of Mixxx it may not be such a big deal to duplicate ControllerEngine::getValue/getParameter and ControllerEngine::setValue/setParameter, but would you want to replicate the [jog wheel API](https://mixxx.org/wiki/doku.php/midi_scripting#scratching_and_jog_wheels) to support the jog wheels of the Kontrol S4 Mk2 or Kontrol S2 Mk2?

Be-ing, 2017-07-09:
[Discussion on Mixxx email list](https://sourceforge.net/p/mixxx/mailman/mixxx-devel/thread/CAKudYbN-%3D%3DnxUuPYWsnPZv5Cw7ASy-UFRRf5BYY78sFSJsq6%2Bw%40mail.gmail.com/#msg35932010)

harryhaaren, 2017-09-15:
Update: The mixxx PR posted hasn't had much feedback. Currently it is still somewhat WIP, particularly in terms of actually providing new value to Mixxx. It does allow mapping of particular controllers using dynamically-recompiled C code with the TCC backend, and provides POC integration of Ctlra library.

The code is stable, and relatively well designed. The UI elements to improve UX of mapping Ctlra based events needs a lot of attention, and my QT skillz are currently not quite up to scratch.. so if somebody reading this has QT skillz and wants to hack Mixxx/Ctlra, do let me know.

In terms of OpenAV's next steps, we're trying to add "new" value to Mixxx, so that means providing extended controller support, and achieving tighter controller integration than was previously possible. Exactly how we achieve that depends on how the unique controller features are exposed from the Ctlra library - which is currenly being prototyped.

All in all, there is progress made on a regular basis towards Ctlra/Mixxx/Hw integration - but it might not be particularly visbible if you're not following the various Ctlra branches.

harryhaaren, 2017-09-17:
Please follow the Mixxx merging of Ctlra code here: https://github.com/mixxxdj/mixxx/pull/1308

closed 2017-09-17 by harryhaaren