openAVproductions/openAV-Ctlra #15: MIDI support

simonvanderveldt, 2017-07-04:
Similar to #14 I think MIDI support would be very beneficial Ctlra.

First of all the majority of hardware controllers currently in use only support MIDI and I'd expect a large amount of future controllers to also just support MIDI.
This means having MIDI supported as a backend would massively increase the amount of people/users to which Ctlra would be useful. This will probably mean more user contributed mappings, i.e. more value for every developer implemting Ctlra.

And having USB HID, MIDI and OSC support in a single library would mean some real benefits to applications who want to support hardware. It would mean the application developers don't have to write (parts) of the hardware support themselves like they would now, where Ctlra would support USB HID, but they have to do support for MIDI and OSC hardware themselves.

simonvanderveldt, 2017-07-04:
@harryhaaren Are there currently any technical issues preventing addition of MIDI backends?
Or is it mainly about determining Ctlra's scope?

harryhaaren, 2017-07-09:
Ctlra's scope mostly - and how much value there is in doing a lot of effort to support MIDI devices that are (probably) already supported by your DAW of choice. I see no technical issues - a POC implementation of MIDI input has been worked on previously, no show-stoppers experienced there.

harryhaaren, 2017-11-03:
I think that with #57 merged, we can close this issue? There is always "more" work to be done - but right now the MIDI devices I have access to work with the implementation, both on input (CC change and Note On/Off events) and feedback to the device (using standard 3-byte MIDI messages, no sys-ex support).

To me the above is the "MPV" or Minimum Viable Product? Let me know if there are objections to closing, otherwise I'll close in a few days :) -H

harryhaaren, 2017-12-30:
Closing this issue as basic MIDI I/O is working. We can open a new issue to improve the MIDI handling if required.

closed 2017-12-30 by harryhaaren