openAVproductions/openAV-Ctlra #18: abstraction for numeric LED displays

Be-ing, 2017-07-08:
Some controllers, namely the Kontrol F1 & Kontrol S4, have 8 segment LED readouts for displaying numbers. It would be helpful if Cltra provided a way for mappings to display numbers with these displays without having to be concerned with which LEDs to light up.

harryhaaren, 2017-07-09:
Good point - yes there should be something. This is kind of a unique one, because its an output only component (no input from the screen). I'll have to think about this and figure out a good solution. I have an F1 here, so I can prototype APIs with it. Will post back here when I have an update - but its not high on the priorities right now (although I do see it as somewhere hopefully in the next month or 2, 3 or .. ; )

harryhaaren, 2017-09-15:
This issue should be resolved similar to https://github.com/openAVproductions/openAV-Ctlra/commit/144db2d521695847a551e025f7abb21b6bb0d6da where a new "Feedback Item" is added for LED strip functionality (aka, digital VU meter lights).

The abstraction of a "numeric LED display" is just a number of LEDs that can be logically grouped to display a number - a convenience API over the already-available-in-ctlra `light_set()` APIs.

I'll leave this issue open, as I currently don't have 7 segment displays working on a controller yet. I do have access to an NI F1, which has a psuedo 3-digit display, which can be used to prototype this functionality.