openAVproductions/openAV-Ctlra #19: support old Native Instruments controllers

Be-ing, 2017-07-09:
Native Instruments' controllers released in the past several years have all been class compliant HID devices. Before that, they used a custom USB protocol. The snd-usb-caiaq driver in Linux makes the manipulation of these controllers accessible as generic Linux input events, but AFAIK no application has ever made use of this because it is not typical to deal with generic Linux input events. However, if these devices were supported by Ctlra in userspace, they could be mapped to Mixxx in a cross-platform way. Would Ctlra's use of libusb for the controller endpoints interfere with ASIO and CoreAudio drivers for the sound cards built into some of these controllers?

My friend has an old Kontrol X1 Mk1 that he does not use anymore, so maybe I could use that to add Ctlra support for it.

See also https://bugs.launchpad.net/mixxx/+bug/1432442

harryhaaren, 2017-07-09:
Interesting - I wasn't aware that the caiaq module presented events, I was under the impression it only provided access to the soundcard via ALSA, and nothing more. There is value in Ctlra supporting *all* NI devices (Mk1 and Mk2), because then any linux-audio user could just be told "get NI gear" and you know it will work, hotplug, and be totally accessible for them.

I know the devices don't have a "MIDI mode" switch - all the translation is done in the driver, and a virtual MIDI device is enabled. This logically makes sense, think about designing and testing 2 firmware modes, or two software modes - which would you choose? :)

A small single .c file that demonstrates how to access each event individually would help greatly - I don't think I'd jump right into creating a Ctlra implementation straight away. That said - your time your call :)

Be-ing, 2017-07-09:
> This logically makes sense, think about designing and testing 2 firmware modes, or two software modes - which would you choose?

Good point, I had not thought of it that way.

> A small single .c file that demonstrates how to access each event individually would help greatly

[Here it is](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/sound/usb/caiaq/input.c?h=v4.12)

harryhaaren, 2017-07-09:
Wait - is that `input.c` part of the kernel module itself to actually achieve the NHL events -> linux event mapping, or is it a demo program to show how to consume the linux events?

By the small single .c file, I meant a userspace demo program how to consume the events (like Ctlra would).

Be-ing, 2017-07-09:
Yes, that file is part of the kernel module. Once upon a time I started a little Python program to translate those generic Linux input events to MIDI for the original Kore controller. I never finished that or published it after realizing that controller didn't work well for DJing anyway. :P I was thinking that Ctlra could access the USB device directly without needing the snd-usb-caiaq driver to translate the USB traffic into Linux input events, but now that I think about it that could interfere with the audio part of the driver. Perhaps the driver could be split into audio and control parts, then users who wanted to use Ctlra could blacklist the control kernel module.

Be-ing, 2017-07-09:
> Perhaps the driver could be split into audio and control parts

Ehh, but we can't do that with the proprietary Mac and Windows drivers. :/ So unfortunately I think this would have to be platform dependent.

Be-ing, 2017-07-09:
For the particular case of the Kontrol X1 Mk1, there would be no issue with the audio driver because the device has no sound card.

marceliq, 2017-07-09:
Hi there, I really like the initiative, because I'm (not proud, because of silly buy, without checking midi support of this device on linux) owner of Kontrol S8. I don't know if you guys have access to it, so if you don't, I might somehow contribute to the project. But I'm not C/C++ programmer :( Just python, but maybe for some debugging it could be enough :)

Back to the issue. S8 also have HID for faders and knobs and sound card is working via caiaq driver, but I could not manage to run it in multi-device mode in JACK.

harryhaaren, 2017-07-09:
Hi @marceliq! Wauw - the S8 is a really nice device.. unfortunately I do not have access to it.. but Ctlra is hungry for S8 support!

As I have experience with the other Kontrol series (S2 and D2), I *think* I know what to expect in terms of what the S8 is capable of, and how it is (probably) mapped.

If you are comfortable compiling Ctlra from source, we can work in a s8 branch, and I can do most of the C coding, and you'll have to spend time to note down the details of the USB HID messages. I'll then iterate over the code, and we can see if that is a workable way of achieving an S8 implementation for Ctlra!?

The immidiate benefit for you would be that the "daemon" example in Ctlra would expose your S8 as a MIDI device in the QJackCtl ALSA MIDI tab, which would allow you to use it as a standard MIDI device on linux..

If you are interested, please open a new issue for "NI Kontrol S8 Support", and we can discuss details there?

Cheers, -Harry

Be-ing, 2017-07-10:
The S8 is one of the newer generation of NI controllers. As a class compliant device, the sound card is handled by the generic snd-usb-audio driver. Perhaps after we have S8 support in Ctlra, maybe you'd be interested in working on adding Remix Decks to Mixxx @marceliq? Otherwise I don't know what you're going to map all those faders to.

harryhaaren, 2017-09-15:
Closing this until somebody has access to hardware, or can provide me with hardware access.

closed 2017-09-15 by harryhaaren

apocalypse2012, 2021-05-12:
I have a Guitar Rig 3 that is both audio and controller. This is likely long dead, but I am still interested in this.

harryhaaren, 2021-05-12:
Hey @apocalypse2012, Thanks for posting your request - this is open-source, anybody can pickup where the project is today and improve :) Can I ask you open a new ticket just for that device, and include the output of `lsusb` when the device is connected? A `sudo usbview` would also be good, giving lots of USB detail. I'm not sure what USB protocols it actually speaks... if its USB HID buttons/switches, chances are it will be somewhat similar to existing Ctlra device code, if its NI proprietary it make the job a lot bigger/more-difficult.

(Re long dead; I don't have/make/spend a lot of time on adding new controllers, but do continue to use the Ctlra library for USB device access, and it works extremely well/reliably on the Linux platform I'm using it with.)