openAVproductions/openAV-Ctlra #21: support Numark NS7 & Numark NS7II

Be-ing, 2017-07-09:
The [Numark NS7](https://www.numark.com/product/ns7) and [NS7II](https://www.numark.com/product/ns7ii) use nonstandard extensions to MIDI that require custom Mac and Windows drivers. No driver was written for Linux. It'd be cool if Ctlra could bring Linux support for these old controllers.

Be-ing, 2017-07-09:
The [Numark NS7III](https://www.numark.com/product/ns7iii) seems more similar to the NV & NVII (#22) than the NS7 & NS7II. I'm guessing it uses standard USB MIDI for most of the controller with a USB endpoint that uses a custom protocol for the screens.

harryhaaren, 2017-09-15:
Closing this until somebody has access to hardware, or can provide me with hardware access.

closed 2017-09-15 by harryhaaren