openAVproductions/openAV-Ctlra #22: support Numark NV & Numark NVII

Be-ing, 2017-07-09:
The [Numark NV](https://www.numark.com/product/nv) and [Numark NVII](https://www.numark.com/product/nvii) are DJ controllers with screens. I'm not sure about the details of how they communicate over USB, but I am guessing they are similar to the Ableton Push 2 (#12), using MIDI for most of the controller and a separate USB endpoint with a custom protocol for the screens.

harryhaaren, 2017-07-09:
The issue is getting access to the hardware. High quality, powerful devices are expensive. If somebody is willing to give me (or another dev) access to the device for some time, we can investigate it.

Interestingly, [this article](https://www.numark.com/kb/article/2253) says that the NVII device is class-compliant, and hence for MacOS it does *not* require a driver. Given I don't have access, there's no way to find out (yet?)..

Be-ing, 2017-07-09:
> The issue is getting access to the hardware.

Yes, I understand. I'm trying to identify which devices Ctlra would be suited for, so if someone who has the hardware is curious and has the required skills, they have an idea where to start.

> Interestingly, this article says that the NVII device is class-compliant, and hence for MacOS it does not require a driver.

My guess is that Serato accesses the USB endpoint for the screens directly. This what Ableton seems to do with the Push 2, which also [does not require the user to install a driver](https://help.ableton.com/hc/en-us/articles/209776005-Setting-up-Push-2-Mac-).

harryhaaren, 2017-09-15:
Closing this until somebody has access to hardware, or can provide me with hardware access.

closed 2017-09-15 by harryhaaren

darkspr1te, 2018-05-07:
Here with Numark NV2 USB Pcap from VDJ8 , am no expert but it seems to use midi commands. Device 1.2.? is the Numark NV2 graphics, 1.3.x is audio and 1.4.x is platter/buttons. 
[numarknv2.pcap.tar.gz](https://github.com/openAVproductions/openAV-Ctlra/files/1979443/numarknv2.pcap.tar.gz)

harryhaaren, 2018-05-11:
Thanks @darkspr1te for the .pcap. I don't see myself having time to work on deciphering things right now, thanks for your input here! I had a quick look at the pcap - yep looks like you're spot on with the graphics endpoints anyway.

Would you provide a lsusb / usbview of the device? I'm curious what usb end-points it exposes, and more interestingly what is exposes them as? USB HID? USB custom vendor? Only USB-Midi sysex?

darkspr1te, 2018-05-16:
Here with the reequested lsusb stuff, I have sent various sysex messages to the GFX interface via midi-sysex app and had random gfx changes like scroll bar show up. 
[numark-usb.txt](https://github.com/openAVproductions/openAV-Ctlra/files/2007818/numark-usb.txt)