openAVproductions/openAV-Ctlra #24: NI Kontrol S8 Support

marceliq, 2017-07-09:
As we discussed in #19 I'd would be nice to have a daemon which expose S8 as a MIDI device and of course implementation for Ctlra. Compiling from source should not be a problem.

harryhaaren, 2017-07-09:
OK Great! Lets do this. I'll write down a few steps, that hopefully will get us started. This will be a long process, and it will require that we "bounce" code and information back and forwards a few times. Its not going to be particularly easy, and I have not written a driver for a device I don't have physical access to before - but lets rise to the challenge!

After pluggin in the device, please paste the output of the following commands.
`lsusb` and `dmesg | tail -n40` and `cat /sys/kernel/debug/usb/devices`  (Put these into seperate sections, with three "backtick" characters to have github format them nicely if possible :)

With that info, I should be able to write some code to dump any incoming USB HID packets, and then we can get started on the process of decoding the bits!

marceliq, 2017-07-09:
I'm realy realy sorry. Currently I'm on a short vacation :( But I'll be back on Wednesday evening.

harryhaaren, 2017-07-09:
No need to be sorry at all. We should both understand this process will take a few weeks (minimum!) so vacation, days off, and such are all very normal! I know I can be pretty enthusiastic (read above ;) but please don't feel any pressure to prioritze this. 

OpenAV is a spare-time project for me - I will prioritize daily things over the S8 driver work - as should you! Enjoy your holidays, I'll get an email when you post here - so lets continue then. Cheers, -Harry

marceliq, 2017-07-12:
Hi, here are the outputs:
`lsusb`
```
Bus 004 Device 002: ID 8087:8000 Intel Corp. 
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:8008 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 003: ID 138a:0017 Validity Sensors, Inc. Fingerprint Reader
Bus 002 Device 006: ID 17cc:1370 Native Instruments 
Bus 002 Device 005: ID 04ca:7035 Lite-On Technology Corp. 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
`dmesg | tail -n40`
```
[13449.597456] e1000e: enp0s25 NIC Link is Down
[13449.600354] IPv6: ADDRCONF(NETDEV_UP): enp0s25: link is not ready
[13449.840890] IPv6: ADDRCONF(NETDEV_UP): enp0s25: link is not ready
[13449.842253] IPv6: ADDRCONF(NETDEV_UP): wlp4s0: link is not ready
[13449.844391] iwlwifi 0000:04:00.0: L1 Enabled - LTR Enabled
[13449.844670] iwlwifi 0000:04:00.0: L1 Enabled - LTR Enabled
[13450.040540] iwlwifi 0000:04:00.0: L1 Enabled - LTR Enabled
[13450.040818] iwlwifi 0000:04:00.0: L1 Enabled - LTR Enabled
[13450.055180] IPv6: ADDRCONF(NETDEV_UP): wlp4s0: link is not ready
[13450.066461] iwlwifi 0000:04:00.0: L1 Enabled - LTR Enabled
[13450.066729] iwlwifi 0000:04:00.0: L1 Enabled - LTR Enabled
[13450.270271] iwlwifi 0000:04:00.0: L1 Enabled - LTR Enabled
[13450.270538] iwlwifi 0000:04:00.0: L1 Enabled - LTR Enabled
[13450.283632] IPv6: ADDRCONF(NETDEV_UP): wlp4s0: link is not ready
[13450.377845] IPv6: ADDRCONF(NETDEV_UP): wlp4s0: link is not ready
[13452.440297] usb 2-7: reset full-speed USB device number 3 using xhci_hcd
[13453.187819] xhci_hcd 0000:00:14.0: WARN Event TRB for slot 1 ep 4 with no TDs queued?
[13453.342318] e1000e: enp0s25 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: Rx/Tx
[13453.342354] IPv6: ADDRCONF(NETDEV_CHANGE): enp0s25: link becomes ready
[13453.642659] IPv6: ADDRCONF(NETDEV_UP): wlp4s0: link is not ready
[13456.699408] wlp4s0: authenticate with e8:94:f6:d0:cb:58
[13456.701915] wlp4s0: send auth to e8:94:f6:d0:cb:58 (try 1/3)
[13456.702526] wlp4s0: authenticated
[13456.709856] wlp4s0: associate with e8:94:f6:d0:cb:58 (try 1/3)
[13456.710737] wlp4s0: RX AssocResp from e8:94:f6:d0:cb:58 (capab=0x11 status=0 aid=2)
[13456.719614] wlp4s0: associated
[13456.719622] IPv6: ADDRCONF(NETDEV_CHANGE): wlp4s0: link becomes ready
[13457.783320] xhci_hcd 0000:00:14.0: WARN Event TRB for slot 1 ep 4 with no TDs queued?
[13457.865386] xhci_hcd 0000:00:14.0: WARN Event TRB for slot 1 ep 4 with no TDs queued?
[13621.740180] usb 2-6: new high-speed USB device number 6 using xhci_hcd
[13621.922166] hid-generic 0003:17CC:1370.0002: hiddev0,hidraw0: USB HID v1.10 Device [Native Instruments  Traktor Kontrol S8] on usb-0000:00:14.0-6/input5
[13621.951750] usbcore: registered new interface driver snd-usb-audio
[13795.560083] usb 2-7: reset full-speed USB device number 3 using xhci_hcd
[13796.324093] xhci_hcd 0000:00:14.0: WARN Event TRB for slot 1 ep 4 with no TDs queued?
[13797.912743] xhci_hcd 0000:00:14.0: WARN Event TRB for slot 1 ep 4 with no TDs queued?
[13797.989358] xhci_hcd 0000:00:14.0: WARN Event TRB for slot 1 ep 4 with no TDs queued?
[13907.133489] usb 2-7: reset full-speed USB device number 3 using xhci_hcd
[13907.886950] xhci_hcd 0000:00:14.0: WARN Event TRB for slot 1 ep 4 with no TDs queued?
[13909.021792] xhci_hcd 0000:00:14.0: WARN Event TRB for slot 1 ep 4 with no TDs queued?
[13909.106972] xhci_hcd 0000:00:14.0: WARN Event TRB for slot 1 ep 4 with no TDs queued?
```
`cat /sys/kernel/debug/usb/devices`
```
T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 3
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 4.11
S:  Manufacturer=Linux 4.11.9-1-ck ehci_hcd
S:  Product=EHCI Host Controller
S:  SerialNumber=0000:00:1a.0
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=01 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  2 Spd=480  MxCh= 6
[... 93 lines cut ...]
S:  SerialNumber=0000:00:14.0
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=04 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 3
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 4.11
S:  Manufacturer=Linux 4.11.9-1-ck ehci_hcd
S:  Product=EHCI Host Controller
S:  SerialNumber=0000:00:1d.0
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=04 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  2 Spd=480  MxCh= 8
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=8087 ProdID=8000 Rev= 0.04
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   2 Ivl=256ms
```
And maybe some important info:
if I do `cat /dev/usb/hiddev0` and move with faders/knobs, I can see "binary data" in terminal.

harryhaaren, 2017-07-13:
OK Great! From the `lsusb` info, we can see the VendorID and ProductID from the `S8: Bus 002 Device 006: ID 17cc:1370 Native Instruments`. The vid is `0x17cc` and pid `0x1370` in hex. This allows reading the next sections better.

`dmesg` gives us 3 interesting lines, informing that the device arrived, that there is a generic USB HID endpoint (as expected), that the device is the S8, and is available as a /dev/ node. Finally, one of the other USB endpoints (more on those later) shows that the audio-interface is automatically recognized by the `snd-usb-audio` kernel module (so it is a standard compliant USB audio interfaces: again as expected : ) 
```
[13621.740180] usb 2-6: new high-speed USB device number 6 using xhci_hcd
[13621.922166] hid-generic 0003:17CC:1370.0002: hiddev0,hidraw0: USB HID v1.10 Device \n
        [Native Instruments  Traktor Kontrol S8] on usb-0000:00:14.0-6/input5
[13621.951750] usbcore: registered new interface driver snd-usb-audio
```

The tricky part to interpret is the USB debug info, because its pretty compact. Firstly, we are not interested in the audio stuff, only in the USB HID components of the S8 for now - the other parts are not required to access (aka, the audio endpoint, the firmware endpoint etc etc). The interesting data pasted below, with some `>> comments` inserted
```
T:  Bus=02 Lev=01 Prnt=01 Port=05 Cnt=01 Dev#=  6 Spd=480  MxCh= 0
D:  Ver= 2.00 Cls=ef(misc ) Sub=02 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=17cc ProdID=1370 Rev= 0.42
S:  Manufacturer=Native Instruments 
S:  Product=Traktor Kontrol S8
S:  SerialNumber=17F2AAFB
C:* #Ifs= 7 Cfg#= 1 Atr=c0 MxPwr=  0mA
>> Ifs (interfaces) is ~= the number of "endpoints" on the device
>> one of these is the USB HID endpoint...
>> "E"ndpoints are marked with an "E" as the first character of a line :)
A:  FirstIf#= 0 IfCount= 4 Cls=01(audio) Sub=00 Prot=20
I:* If#= 0 Alt= 0 #EPs= 0 Cls=01(audio) Sub=01 Prot=20 Driver=snd-usb-audio
I:* If#= 1 Alt= 0 #EPs= 0 Cls=01(audio) Sub=02 Prot=20 Driver=snd-usb-audio
I:  If#= 1 Alt= 1 #EPs= 2 Cls=01(audio) Sub=02 Prot=20 Driver=snd-usb-audio
E:  Ad=01(O) Atr=05(Isoc) MxPS= 112 Ivl=125us
E:  Ad=81(I) Atr=11(Isoc) MxPS=   4 Ivl=1ms
I:* If#= 2 Alt= 0 #EPs= 0 Cls=01(audio) Sub=02 Prot=20 Driver=snd-usb-audio
I:  If#= 2 Alt= 1 #EPs= 1 Cls=01(audio) Sub=02 Prot=20 Driver=snd-usb-audio
E:  Ad=82(I) Atr=05(Isoc) MxPS= 280 Ivl=125us
I:* If#= 3 Alt= 0 #EPs= 2 Cls=01(audio) Sub=03 Prot=00 Driver=snd-usb-audio
>> above here is all "Driver=snd-usb-audio", so we know its for audio, no HID
>> the next section might be of interest though...

E:  Ad=02(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=83(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
>> So these endpoints are most-likely to address the screen.
>> Given the data-rates for screens, they often use "Bulk" endpoints
>> Still not the buttons / faders though...

>> Ah here we have the buttons/faders;
>> interfaces marked with "Class 03 HID"
I:* If#= 4 Alt= 0 #EPs= 0 Cls=fe(app. ) Sub=01 Prot=01 Driver=(none)
I:* If#= 5 Alt= 0 #EPs= 2 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbhid
E:  Ad=84(I) Atr=03(Int.) MxPS=  64 Ivl=500us
E:  Ad=03(O) Atr=03(Int.) MxPS=  64 Ivl=1ms
>> "Int" is interrupt - the mode that seems to be used by
>> these devices to inform the PC of incoming events
>> The data-rate itself (500us) also indicates that the messages
>> can occur very frequently on AD=84(i)
>> and not quite as frequencly on Ad=03(O)

I:* If#= 6 Alt= 0 #EPs= 1 Cls=ff(vend.) Sub=bd Prot=00 Driver=(none)
E:  Ad=04(O) Atr=02(Bulk) MxPS= 512 Ivl=125us
>> This bulk endpoint is "output", probably to push new firmware to
>> the device. Its a bulk endpoint, and marked as "Class 0xFF Vendor".
```

OK - step 1 is done; initial investigation of the USB endpoints. Next step is for me to write some code, and to try get the "binary data" you refer to printing from a Ctlra application. This requires a little trickery because I obviously can't test the code here - so it might take a few attempts to get it right.

For now, I'll get preparing the code, and setting it up. I'll post here again once I have a branch pushed to Github with the initial framework of the S8 driver, and then we can work together to compile it and test it on your machine.

That's all for now!

harryhaaren, 2017-07-13:
I've duplicated the D2 driver and renamed to S8, but I do not expect anything useful from it yet. I will ping you here when there is something worth testing :)

harryhaaren, 2017-07-14:
Hi @marceliq! The two commits I just pushed *should* print out the data of the S8, when you touch / move things. As before - I'm working blindly here, testing with another device to ensure the code works as expected, but I obviously can't test with the S8 device!

Please compile the S8 branch from git, and run the "build/examples/simple" binary. It should print `ctlra_ni_kontrol_s8_connect : Connected` if the USB HID connection was successful. If that happens, try moving some controls, and see if the USB HID messages are printed on the terminal.

Once I have a report back if we're good so far, we can start looking at decoding a button push-event! Cheers, -Harry

marceliq, 2017-07-14:
Hi @harryhaaren :) I just gave it a try, but it is not working as expected :(

build/examples/simple 
```
ctlra_ni_kontrol_s8_connect: Connected!
request for handle beyond available iface per dev range
ctlra_ni_kontrol_s8_connect: failed to open button usb interface
```
Any chance I might be helpful with that?

harryhaaren, 2017-07-14:
OK no problem - there are going to be a few of these due to not working with the device directly. There is a program called `usbview`, apt-get install should just get it for you. Would you run it, select the S8 device on the left (probably has red-writing) and then post a screenshot?

The program gives the same USB output, but in a much easier to read way. That will let me double-check the values I'm using for USB interfaces :)

Note that the first "Connected!" is already a success, so 50% aint' bad!

marceliq, 2017-07-14:
Here's the output of usbview:
```
Traktor Kontrol S8
Manufacturer: Native Instruments 
Serial Number: 17F2AAFB
Speed: 480Mb/s (high)
USB Version:  2.00
Device Class: ef(misc )
Device Subclass: 02
Device Protocol: 01
Maximum Default Endpoint Size: 64
Number of Configurations: 1
Vendor Id: 17cc
Product Id: 1370
Revision Number:  0.42

Config Number: 1
	Number of Interfaces: 7
	Attributes: c0
	MaxPower Needed:   0mA

	Interface Number: 0
		Name: snd-usb-audio
		Alternate Number: 0
		Class: 01(audio) 
		Sub Class: 01
		Protocol: 20
		Number of Endpoints: 0

	Interface Number: 1
		Name: snd-usb-audio
		Alternate Number: 0
		Class: 01(audio) 
		Sub Class: 02
		Protocol: 20
		Number of Endpoints: 0

	Interface Number: 1
		Name: snd-usb-audio
		Alternate Number: 1
		Class: 01(audio) 
		Sub Class: 02
		Protocol: 20
		Number of Endpoints: 2

			Endpoint Address: 01
			Direction: out
			Attribute: 5
			Type: Isoc
			Max Packet Size: 112
			Interval: 125us

			Endpoint Address: 81
			Direction: in
			Attribute: 17
			Type: Isoc
			Max Packet Size: 4
			Interval: 1ms

	Interface Number: 2
		Name: snd-usb-audio
		Alternate Number: 0
		Class: 01(audio) 
		Sub Class: 02
		Protocol: 20
		Number of Endpoints: 0

	Interface Number: 2
		Name: snd-usb-audio
		Alternate Number: 1
		Class: 01(audio) 
		Sub Class: 02
		Protocol: 20
		Number of Endpoints: 1

			Endpoint Address: 82
			Direction: in
			Attribute: 5
			Type: Isoc
			Max Packet Size: 280
			Interval: 125us

	Interface Number: 3
		Name: snd-usb-audio
		Alternate Number: 0
		Class: 01(audio) 
		Sub Class: 03
		Protocol: 00
		Number of Endpoints: 2

			Endpoint Address: 02
			Direction: out
			Attribute: 2
			Type: Bulk
			Max Packet Size: 512
			Interval: 0ms

			Endpoint Address: 83
			Direction: in
			Attribute: 2
			Type: Bulk
			Max Packet Size: 512
			Interval: 0ms

	Interface Number: 4
		Name: (none)
		Alternate Number: 0
		Class: fe(app. ) 
		Sub Class: 01
		Protocol: 01
		Number of Endpoints: 0

	Interface Number: 5
		Name: usbhid
		Alternate Number: 0
		Class: 03(HID  ) 
		Sub Class: 00
		Protocol: 00
		Number of Endpoints: 2

			Endpoint Address: 84
			Direction: in
			Attribute: 3
			Type: Int.
			Max Packet Size: 64
			Interval: 500us

			Endpoint Address: 03
			Direction: out
			Attribute: 3
			Type: Int.
			Max Packet Size: 64
			Interval: 1ms

	Interface Number: 6
		Name: (none)
		Alternate Number: 0
		Class: ff(vend.) 
		Sub Class: bd
		Protocol: 00
		Number of Endpoints: 1

			Endpoint Address: 04
			Direction: out
			Attribute: 2
			Type: Bulk
			Max Packet Size: 512
			Interval: 125us
```

marceliq, 2017-07-14:
And here's the output:
```
8087:8000 (serial 0) (bus 2, device 2) path: 1
1d6b:0002 (serial 1) (bus 2, device 1)
8087:8008 (serial 0) (bus 1, device 2) path: 1
1d6b:0002 (serial 1) (bus 1, device 1)
1d6b:0003 (serial 1) (bus 4, device 1)
138a:0017 (serial 1) (bus 3, device 3) path: 7
17cc:1370 (serial 2) (bus 3, device 14) path: 6
ctlra_ni_kontrol_s8_connect: Connected! usb_open() success
ctlra_ni_kontrol_s8_connect: Connected! open_interface success
ctlra_ni_kontrol_s8_connect: Connected! connect() returning 0x7f4fe94b0010
simple: accepting Native Instruments Kontrol S8
8087:8000 (serial 0) (bus 2, device 2) path: 1
1d6b:0002 (serial 1) (bus 2, device 1)
8087:8008 (serial 0) (bus 1, device 2) path: 1
1d6b:0002 (serial 1) (bus 1, device 1)
1d6b:0003 (serial 1) (bus 4, device 1)
138a:0017 (serial 1) (bus 3, device 3) path: 7
17cc:1370 (serial 2) (bus 3, device 14) path: 6
04ca:7035 (serial 0) (bus 3, device 5) path: 12
1d6b:0002 (serial 1) (bus 3, device 1)
8087:8000 (serial 0) (bus 2, device 2) path: 1
1d6b:0002 (serial 1) (bus 2, device 1)
8087:8008 (serial 0) (bus 1, device 2) path: 1
1d6b:0002 (serial 1) (bus 1, device 1)
1d6b:0003 (serial 1) (bus 4, device 1)
138a:0017 (serial 1) (bus 3, device 3) path: 7
17cc:1370 (serial 2) (bus 3, device 14) path: 6
04ca:7035 (serial 0) (bus 3, device 5) path: 12
1d6b:0002 (serial 1) (bus 3, device 1)
8087:8000 (serial 0) (bus 2, device 2) path: 1
1d6b:0002 (serial 1) (bus 2, device 1)
8087:8008 (serial 0) (bus 1, device 2) path: 1
1d6b:0002 (serial 1) (bus 1, device 1)
1d6b:0003 (serial 1) (bus 4, device 1)
138a:0017 (serial 1) (bus 3, device 3) path: 7
17cc:1370 (serial 2) (bus 3, device 14) path: 6
04ca:7035 (serial 0) (bus 3, device 5) path: 12
1d6b:0002 (serial 1) (bus 3, device 1)
8087:8000 (serial 0) (bus 2, device 2) path: 1
1d6b:0002 (serial 1) (bus 2, device 1)
8087:8008 (serial 0) (bus 1, device 2) path: 1
1d6b:0002 (serial 1) (bus 1, device 1)
1d6b:0003 (serial 1) (bus 4, device 1)
138a:0017 (serial 1) (bus 3, device 3) path: 7
17cc:1370 (serial 2) (bus 3, device 14) path: 6
04ca:7035 (serial 0) (bus 3, device 5) path: 12
1d6b:0002 (serial 1) (bus 3, device 1)
```

marceliq, 2017-07-14:
left play button - push:
```
hid read () got 41
ni_kontrol_s8_usb_read_cb : size = 41
01 ff ff ff 00 00 00 00 00 10 00 00 00 00 00 40 00 00 00 00 20 00 00 00 5a 00 a0 97 00 00 00 00 9f 97 00 00 00 00 00 00 0e 
hid read () got 41
```
left play button - release:
```
ni_kontrol_s8_usb_read_cb : size = 41
01 ff ff ff 00 00 00 00 00 00 00 00 00 00 00 40 00 00 00 00 20 00 00 00 5a 00 06 9f 00 00 00 00 05 9f 00 00 00 00 00 00 0e 
```
crossfader:
```
hid read () got 109
ni_kontrol_s8_usb_read_cb : size = 109
02 75 00 b9 02 69 00 b2 00 1d 01 aa 00 a3 00 7e 03 09 00 09 00 09 00 09 00 0e 00 0d 00 0d 00 0e 00 12 00 85 00 12 00 0d 00 ee 0f e8 07 37 08 b3 07 f2 07 2a 08 e6 07 3f 08 e3 07 11 08 80 07 e9 07 c6 07 ff 0f ec 07 dc 07 03 08 12 08 ea 07 fe 07 fa 07 d2 07 ec 07 c5 07 62 07 b5 08 04 08 ef 07 a5 07 95 07 ef 07 d4 07 ed 07 e7 07 
hid read () got 109
ni_kontrol_s8_usb_read_cb : size = 109
02 75 00 b9 02 69 00 b2 00 1d 01 aa 00 a3 00 7e 03 09 00 09 00 09 00 09 00 0e 00 0d 00 0d 00 0e 00 12 00 85 00 12 00 0d 00 e6 0f e8 07 37 08 b3 07 f2 07 2a 08 e6 07 3f 08 e3 07 11 08 80 07 e9 07 c6 07 ff 0f ec 07 dc 07 03 08 12 08 ea 07 fe 07 fa 07 d2 07 ec 07 c5 07 62 07 b5 08 04 08 ef 07 a5 07 95 07 ef 07 d4 07 ed 07 e7 07 
hid read () got 109
ni_kontrol_s8_usb_read_cb : size = 109
02 75 00 b9 02 69 00 b2 00 1d 01 aa 00 a3 00 7e 03 09 00 09 00 09 00 09 00 0e 00 0d 00 0d 00 0e 00 12 00 85 00 12 00 0d 00 de 0f e8 07 37 08 b3 07 f2 07 2a 08 e6 07 3f 08 e3 07 11 08 80 07 e9 07 c6 07 ff 0f ec 07 dc 07 03 08 12 08 ea 07 fe 07 fa 07 d2 07 ec 07 c5 07 62 07 b5 08 04 08 ef 07 a5 07 95 07 ef 07 d4 07 ed 07 e7 07 
hid read () got 109
ni_kontrol_s8_usb_read_cb : size = 109
02 75 00 b9 02 69 00 b2 00 1d 01 aa 00 a3 00 7e 03 09 00 09 00 09 00 09 00 0e 00 0d 00 0d 00 0e 00 12 00 85 00 12 00 0d 00 d6 0f e8 07 37 08 b3 07 f2 07 2a 08 e6 07 3f 08 e3 07 11 08 80 07 e9 07 c6 07 ff 0f ec 07 dc 07 03 08 12 08 ea 07 fe 07 fa 07 d2 07 ec 07 c5 07 62 07 b5 08 04 08 ef 07 a5 07 95 07 ef 07 d4 07 ed 07 e7 07 
hid read () got 109
ni_kontrol_s8_usb_read_cb : size = 109
02 75 00 b9 02 69 00 b2 00 1d 01 aa 00 a3 00 7e 03 09 00 09 00 09 00 09 00 0e 00 0d 00 0d 00 0e 00 12 00 85 00 12 00 0d 00 ce 0f e8 07 37 08 b3 07 f2 07 2a 08 e6 07 3f 08 e3 07 11 08 80 07 e9 07 c6 07 ff 0f ec 07 dc 07 03 08 12 08 ea 07 fe 07 fa 07 d2 07 ec 07 c5 07 62 07 b5 08 04 08 ef 07 a5 07 95 07 ef 07 d4 07 ed 07 e7 07 
hid read () got 109
ni_kontrol_s8_usb_read_cb : size = 109
02 75 00 b9 02 69 00 b2 00 1d 01 aa 00 a3 00 7e 03 09 00 09 00 09 00 09 00 0e 00 0d 00 0d 00 0e 00 12 00 85 00 12 00 0d 00 c6 0f e8 07 37 08 b3 07 f2 07 2a 08 e6 07 3f 08 e3 07 11 08 80 07 e9 07 c6 07 ff 0f ec 07 dc 07 03 08 12 08 ea 07 fe 07 fa 07 d2 07 ec 07 c5 07 62 07 b5 08 04 08 ef 07 a5 07 95 07 ef 07 d4 07 ed 07 e7 07 
```

harryhaaren, 2017-07-14:
Cool, nice work on testing! We're making pretty good progress - admittely a rough start with libUSB issues, but with Linux `/dev/hidrawX` access, we have HID packet data! Decoding the data is a long task - for another day! So far so good :)

harryhaaren, 2017-09-15:
Hey @marceliq - open question - do you see yourself finding time to work on this with the S8 in the near-future? I'll close this for now if not. (We can re-open when there is activity again)

harryhaaren, 2017-09-17:
Closing for now. Ctlra is still interested in supporting the S8 device - but it requires help from the community (from people with access to the hardware) or else OpenAV to get access to this hardware for an amount of time. Until either of those happen, this can remain closed.

andersson82, 2019-03-22:
Hi!
I own a S8, and I have tried getting OpenAV-Ctlra to work in Debian today, without sucess, to see if I perhaps could help.
Can you give me instructions on how to build ctrla and I will try my best to support you into getting it to work.
When I follow the build-instructions it looks like it wants AVTK, but dont know where to place the lib that is created when I build AVTK. (Cloned from github)

harryhaaren, 2019-03-23:
Hey @andersson82, thanks for posting! AVTK is a UI library used by OpenAV for various audio UIs (eg, ArtyFX plugins), its available here; https://github.com/openAVproductions/openAV-AVTK.

To be clear - supporting a new device is difficult, and takes a lot of time and patience - I'm happy to try help as best I can (keep in mind that I have only limited time too! : )

I've been working on a more generic method for creating device-support in Ctlra, I think with the S5 and S8 support being suggested/helped by community, maybe I need to up that to try work with.. it should be more usable than the current C code method...

Let me know if you get the AVTK dependency fixed, and we can see how S8 supports look then. Cheers, -Harry

prime-optimal, 2026-08-31:
Is anybody still alive?  I recently got an S8 and have been scouring the internet for a way to get the screens working in Djay Pro.  Seems like it's possible to do it with the D2 in Mixxx so that gives me hope.  

Going to try to get what exists working on my machine, as I also have an F1 and a Maschine MK2