openAVproductions/openAV-Ctlra #29: NI Kontrol S2 Mk1 support

Be-ing, 2017-07-31:
I have been helping @turlando use the JS HID packet parser library in Mixxx to map the NI Kontrol S2 Mk1. @turlando reported that the USB product ID for the Mk1 is 0x1101 whereas the product ID used for the current S2 support in Ctlra is 0x1320. The existing support should be renamed to clarify that it is for the S2 Mk2, not Mk1.

harryhaaren, 2017-07-31:
@Be-ing; correct on the S2 MK2, will update thanks.

On filing issues for every controller out there - unless somebody has access (or can get me access) to a device, there is little point filing an issue. 99.99% of controllers could have a "new device request" issue here without any progress..

Be-ing, 2017-07-31:
@turlando, would you be interested in implementing this?

harryhaaren, 2017-07-31:
As far as I understand, the MK1 is NHL (Native Hardware Language) and not plain-old USB HID. That means that unless you're planning on figuring out NHL in Mixxx's JS HID I don't see how it will work? The existing kernel module for NIs stuff exposes "linux events" for the devices - it should be possible to figure out how to grab the data using the kernel module though..

Be-ing, 2017-07-31:
I thought so too, but apparently it is an HID device. @turlando has it starting to work with Mixxx's existing HID system. [NI's help article](https://support.native-instruments.com/hc/en-us/articles/210293705) just lists "S2" as class compliant without specifying Mk1 or Mk2, so I guess they both are.

harryhaaren, 2017-07-31:
Ah ok - in that case, @turlando you might want to try using the latest Ctlra to see if the USB HID messages match up.. might save you a few hours / days work! Just updating the USB VID / PID in the S2mk2 driver should probably do..?

harryhaaren, 2017-09-15:
Closing this until somebody has access to hardware, or can provide me with hardware access.

closed 2017-09-15 by harryhaaren

harryhaaren, 2019-01-27:
@ronlaws86 has posted USBView info on the S2 Mk1 in the (now only) S4 Mk1 thread;
https://github.com/openAVproductions/openAV-Ctlra/issues/94#issuecomment-455901841

Interface info
```
Interface Number: 0
	Name: snd-usb-audio
	Alternate Number: 0
	Class: 01(audio) 
	Sub Class: 01
	Protocol: 20
	Number of Endpoints: 0

Interface Number: 1
	Name: snd-usb-audio
	Alternate Number: 0
	Class: 01(audio) 
	Sub Class: 02
	Protocol: 20
	Number of Endpoints: 0

Interface Number: 1
	Name: snd-usb-audio
	Alternate Number: 1
	Class: 01(audio) 
	Sub Class: 02
	Protocol: 20
	Number of Endpoints: 2

		Endpoint Address: 01
		Direction: out
		Attribute: 5
		Type: Isoc
		Max Packet Size: 240
		Interval: 125us

		Endpoint Address: 81
		Direction: in
		Attribute: 17
		Type: Isoc
		Max Packet Size: 4
		Interval: 1ms

Interface Number: 2
	Name: (none)
	Alternate Number: 0
	Class: fe(app. ) 
	Sub Class: 01
	Protocol: 01
	Number of Endpoints: 0

Interface Number: 3
	Name: usbhid
	Alternate Number: 0
	Class: 03(HID  ) 
	Sub Class: 00
	Protocol: 00
	Number of Endpoints: 2

		Endpoint Address: 82
		Direction: in
		Attribute: 3
		Type: Int.
		Max Packet Size: 64
		Interval: 125us

		Endpoint Address: 02
		Direction: out
		Attribute: 3
		Type: Int.
		Max Packet Size: 64
		Interval: 125us
```