openAVproductions/openAV-Ctlra #82: Ctlra Mapping: Dynamic User Input Map

harryhaaren, 2018-06-11:
This task involves implementing the dynamic parsing of the ctlra hardware metadata, the software "targets" metadata, and being able to present a simple text-file / UI to allow a user to dynamically remap a control.

TODO
- think about layered mappings here - how to expose to the user?

Be-ing, 2018-06-11:
I think if you make this part of Ctlra's scope, then there should be a way for the application/user to change the layering logic without recompiling Ctlra. If that is hardcoded into the device's Ctlra implementation, it would be like baking it into the device's firmware, [which I think is a bad idea](https://mixxx.org/forums/viewtopic.php?p=38644#p38644).

harryhaaren, 2018-06-11:
Hey @Be-ing - totally agree, nothing hard-baked in here. The idea is to provide the user the flexibility to map the layers, there is no "layering" happening at the device level - the user is in control. Thanks for the input!

Be-ing, 2018-06-11:
Good to hear. While it may be helpful to let device implementations specify default layering logic, applications should have the option to disable that and define their own.

harryhaaren, 2018-06-11:
Yes agreed again. This "mapping" layer will be optional, and exists above the Ctlra generic event layer. I'm hoping to do up some diagrams, and I think that'll put your concerns to rest :)