openAVproductions/openAV-Ctlra #98: Kontrol S5 Support

harryhaaren, 2019-03-16:
This issue tracks support for the Native Instruments S5 controller. To start, we'll need to get some basic USB info for the device; Run usbview as below, then select the "Tracktor Kontrol S5" device, and copy paste the resulting info into a comment below;
```
$ sudo usbview
```

bboissonneault, 2019-03-19:
here's what I got...

Traktor Kontrol S5
Manufacturer: Native Instruments 
Serial Number: 191A8286
Speed: 480Mb/s (high)
USB Version:  2.00
Device Class: ef(misc )
Device Subclass: 02
Device Protocol: 01
Maximum Default Endpoint Size: 64
Number of Configurations: 1
Vendor Id: 17cc
Product Id: 1420
Revision Number:  0.22

Config Number: 1
	Number of Interfaces: 6
	Attributes: c0
	MaxPower Needed:   0mA

	Interface Number: 0
		Name: snd-usb-audio
		Alternate Number: 0
		Class: 01(audio) 
		Sub Class: 01
		Protocol: 20
		Number of Endpoints: 0

	Interface Number: 1
		Name: snd-usb-audio
		Alternate Number: 0
		Class: 01(audio) 
		Sub Class: 02
		Protocol: 20
		Number of Endpoints: 0

	Interface Number: 1
		Name: snd-usb-audio
		Alternate Number: 1
		Class: 01(audio) 
		Sub Class: 02
		Protocol: 20
		Number of Endpoints: 2

			Endpoint Address: 01
			Direction: out
			Attribute: 5
			Type: Isoc
			Max Packet Size: 112
			Interval: 125us

			Endpoint Address: 81
			Direction: in
			Attribute: 17
			Type: Isoc
			Max Packet Size: 4
			Interval: 1ms

	Interface Number: 2
		Name: snd-usb-audio
		Alternate Number: 0
		Class: 01(audio) 
		Sub Class: 02
		Protocol: 20
		Number of Endpoints: 0

	Interface Number: 2
		Name: snd-usb-audio
		Alternate Number: 1
		Class: 01(audio) 
		Sub Class: 02
		Protocol: 20
		Number of Endpoints: 1

			Endpoint Address: 82
			Direction: in
			Attribute: 5
			Type: Isoc
			Max Packet Size: 112
			Interval: 125us

	Interface Number: 3
		Name: usbhid
		Alternate Number: 0
		Class: 03(HID  ) 
		Sub Class: 00
		Protocol: 00
		Number of Endpoints: 2

			Endpoint Address: 83
			Direction: in
			Attribute: 3
			Type: Int.
			Max Packet Size: 64
			Interval: 500us

			Endpoint Address: 02
			Direction: out
			Attribute: 3
			Type: Int.
			Max Packet Size: 64
			Interval: 1ms

	Interface Number: 4
		Name: (none)
		Alternate Number: 0
		Class: ff(vend.) 
		Sub Class: bd
		Protocol: 00
		Number of Endpoints: 1

			Endpoint Address: 03
			Direction: out
			Attribute: 2
			Type: Bulk
			Max Packet Size: 512
			Interval: 125us

	Interface Number: 5
		Name: (none)
		Alternate Number: 0
		Class: fe(app. ) 
		Sub Class: 01
		Protocol: 01
		Number of Endpoints: 0

harryhaaren, 2019-03-20:
Thanks, that pretty much the expected output for the S5 device;
- Interface 3 is USB HID, which provides event input (button, slider, etc), and LED output
- Interface 4 is USB Bulk, with "vendor specific" protocol to operate the screen

The other endpoints are used by A) the kernel for Audio I/O, and interface 5 is for updating the firmware (so we don't go near it :)

That's good input - I'll try find some time at the weekend to get some boilerplate code started for supporting this device. Once that's in place, I'll be asking you to press every button, and write down some details about them - then we can get those working. Sliders would be the next step, then encoders, touch strips, LEDs and screens... lots of work - so lets take small steps in the right direction :)

bboissonneault, 2019-03-21:
Sounds great. I'm ready

harryhaaren, 2019-04-06:
Hi @bboissonneault! It took a bit longer than expected, but we have a commit that hopefully works with the device you have. It would be great if you can test the `examples/simple` application, which should just turn on all LEDs when you start it, and turn (most of) them off again when the application is stopped. The screens should display coloured squares - and go grey/black on shutdown - can you confirm this is actually happening, on both screens? Thanks!

sandersantema, 2019-05-19:
Hi @harryhaaren I've got an S5 and I would be very interested in it being supported by CTRL-A and Mixxx, what can I do to help you? I've got access to macOS and a manjaro system, considering usbview doesn't seem to be a thing on macOS I suppose I'd need to run the aforementioned tests on Linux?

harryhaaren, 2019-05-25:
Hey @sandersantema, thanks for posting up your interest. There's a branch on this github page that has basic support for the S5 ( https://github.com/openAVproductions/openAV-Ctlra/tree/ni_s5 ). If you're familiar with Git, `git checkout ni_s5` should get you that copy of the code for Ctlra.

Regarding support, I don't have a MacOS system to test, so to date Linux is the only supported platform (Other platforms can be easily supported as Ctlra was designed to be cross-platform, however I just don't have access to Mac/Win them to test it). Hence, yes using your Manjaro system is the best idea.

No need for lsusb output, as I now have access to an S5 here myself. If you can test the 2nd screen works that would be great, as the 2nd screen seems to have an issue on my device. (Googling shows its a somewhat common issue that these devices have....)

To get Mixxx & Ctlra working together is a bit more work - but also possible. Let me know how you get on with building Ctlra on Manjaro and we can take it from there! Regards, -Harry

harryhaaren, 2020-01-07:
Branch `ni_s5` has a few fixes from just now, touchstrip leds and right screen drawing fixes; https://github.com/openAVproductions/openAV-Ctlra/tree/ni_s5