openAVproductions/openAV-Ctlra #105: Run-time dependency openav_avtka NOT found

Luxcium, 2020-10-11:
I am getting an error ***Run-time dependency openav_avtka found: NO (tried pkgconfig and cmake)***

and a warning ***meson.build:55: WARNING: Got an empty configuration_data() object and found no substitutions in the input file 'loopa_mk3.c'. If you want to copy a file to the build dir, use the '***

I have built openav_avtka (but I dont know how to install it or link it into openAV-Ctlra

```
The Meson build system
Version: 0.55.3
Source dir: /home/luxcium/.local/src/openAV-Ctlra
Build dir: /home/luxcium/.local/src/openAV-Ctlra/build
Build type: native build
Project name: openav_ctlra
Project version: undefined
C compiler for the host machine: ccache cc (gcc 10.2.1 "cc (GCC) 10.2.1 20200723 (Red Hat 10.2.1-1)")
C linker for the host machine: cc ld.bfd 2.34-5
Host machine cpu family: x86_64
Host machine cpu: x86_64
Found pkg-config: /usr/bin/pkg-config (1.6.3)
Found CMake: /usr/bin/cmake (3.17.4)
Run-time dependency openav_avtka found: NO (tried pkgconfig and cmake)

|Executing subproject avtka method meson 
|
|Project name: openav_avtka
|Project version: undefined
|C compiler for the host machine: ccache cc (gcc 10.2.1 "cc (GCC) 10.2.1 20200723 (Red Hat 10.2.1-1)")
|C linker for the host machine: cc ld.bfd 2.34-5
|C++ compiler for the host machine: ccache c++ (gcc 10.2.1 "c++ (GCC) 10.2.1 20200723 (Red Hat 10.2.1-1)")
|C++ linker for the host machine: c++ ld.bfd 2.34-5
|Run-time dependency cairo found: YES 1.16.0
|Run-time dependency x11 found: YES 1.6.12
|Build targets in project: 1
|Subproject avtka finished.

Has header "libtcc.h" : YES 
Run-time dependency alsa found: YES 1.2.3.2
Dependency cairo found: YES 1.16.0 (cached)
Run-time dependency jack found: YES 1.9.14
Run-time dependency libusb-1.0 found: YES 1.0.23
Run-time dependency gl found: YES 1.2
Configuring config.h using configuration
Configuring loopa_mk3.c using configuration
meson.build:55: WARNING: Got an empty configuration_data() object and found no substitutions in the input file 'loopa_mk3.c'. If you want to copy a file to the build dir, use the 'copy:' keyword argument added in 0.47.0
Configuring config.h using configuration
Build targets in project: 3

openav_ctlra undefined

  Subprojects
    avtka: YES

Found ninja-1.10.1 at /usr/bin/ninja
```

Luxcium, 2020-10-11:
**`ninja`** →
```
[12/25] Compiling C object ctlra/libctlra_devices.a.p/devices_avtka.c.o
FAILED: ctlra/libctlra_devices.a.p/devices_avtka.c.o 
ccache cc -Ictlra/libctlra_devices.a.p -Ictlra -I../ctlra -I../subprojects/avtka/include -I../subprojects/avtka/src -I/usr/include/libusb-1.0 -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -std=gnu99 -g -DHAVE_AVTKA -fPIC -Wno-unused-variable -MD -MQ ctlra/libctlra_devices.a.p/devices_avtka.c.o -MF ctlra/libctlra_devices.a.p/devices_avtka.c.o.d -o ctlra/libctlra_devices.a.p/devices_avtka.c.o -c ../ctlra/devices/avtka.c
../ctlra/devices/avtka.c: Dans la fonction « ctlra_build_avtka_ui »:
../ctlra/devices/avtka.c:536:6: erreur: « struct avtka_screen_opts_t » n'a pas de membre nommé « flags_rgb »
  536 |     .flags_rgb = 1,
      |      ^~~~~~~~~
[22/25] Compiling C object subprojects/avtka/libavtka.a.p/src_pugl_pugl_x11.c.o
ninja: build stopped: subcommand failed.
```

MrOneTwo, 2020-12-29:
@Luxcium A simple fix to get it to compile is to comment out that `.flags_rgb = 1` line.

harryhaaren, 2022-12-13:
Doh - no idea how I missed this, sorry folks. Yes this is a compile-time issue, fixing now..

closed 2025-06-23 by Luxcium